realtimeradio/ata_snap #7: 4 bit packing

siemion, 2020-10-09:
The attached PDF shows how to pack 4-bit complex values in such a way that the process of unpacking them into 8-bit values can be performed for multiple values at one time.  This works by pretending that the 4 bit values occupy the upper 4 bits of an 8-bit value with zeros in the lower 4 bits, then shifting and bitwise OR-ing successive blocks of bits to interleave them. The diagram uses a block size of 128 bits (16 bytes).  This works well with the 128 bit registers and SSE instructions of modern processors and the 16 byte size corresponds to a complete cache line size.  Larger block sizes such as 256 bits (AVX) or even 512 bits (AVX512) could also be used, but I suspect the performance boost from SSE to AVX(512) would not be as significant as the initial performance gain from non-SSE to SSE.  I also think that a 128 bit block size would be somewhat simpler to implement on the FPGA, though the FGPA implementation is very basic and low cost for any of them.  I think any block size would be way better than placing real and imaginary components in the same byte in the packed format.

It would be great if you could adapt the voltage output mode of the ATA SNAP design to pack the 4 bit real and imaginary components into the packet using this interleaved scheme.  My preference would be for block size of 128 or 256 as the AVX512 instruction set is not so widespread yet.  I think we talked about making the packing mode runtime selectable between 4+4 bit complex bytes (the existing mode) and the interleaved block mode described here, with the mode used for a given packet indicated by a bit in the packet header.  That would be fine with me, but if I had to choose only one mode it would be the interleaved mode.

[4bit_pack_unpack.pdf](https://github.com/realtimeradio/ata_snap/files/5354781/4bit_pack_unpack.pdf)

realtimeradio, 2020-10-09:
Also convert data order to:

(slowest) [chan][time][pol] (fastest)

...rather than the current...

(slowest) [time][chan][pol] (fastest)

siemion, 2020-10-09:
@david-macmahon Can you see this?

david-macmahon, 2020-10-09:
I couldn't have said it better myself! :)