realtimeradio/ata_snap #10: Nov 15 Specifications

siemion, 2020-10-12:
Baseline specs for Nov 15: The Allen Telescope Array Synoptic All Sky SETI Survey: A pi/3 Steradian Pilot

4096 complex channels over 1000 MHz Nyquist band (2 Gsps)
8 tap PFB
Dual 10 GbE output
2 output modes: 
up to 1000 MHz BW @ 4 bits (per polarization)
up to 500 MHz BW @ 8 bits (per polarization)
(16 Gbps)

Channel selection:
contiguous w/ start/stop boundaries (subject to above BW limits for 4/8 bit modes)

Coarse sample delay per input: 512 samples, controllable via SW register.

Ethernet output (baseline):
4 bit mode:
16 destinations w/ 8192 byte payload packets (4096 channels, 256 channels per destination)
32 destinations w/ 4096 byte payload packets (4096 channels, 128 channels per destination)

8 bit mode:
32 destinations w/ 8192 byte payload packets (4096 channels, 128 channels per destination)
64 destinations w/ 4096 byte payload packets (4096 channels, 64 channels per destination)

Packet format:
64 bit header + 8192 bytes data
For N channels and M time steps:
c0p0t0_r, c0p0t0_i, c0p1t0_r, c0p1t0_i, c1p0t0_r, .... cNp1tM_i    
or in other words, M time ordered complex spectra of N channels.
From fastest to slowest changing dimensions:
polarization, channel, time

Ethernet output (stretch goal):
4 bit mode:
64 destinations w/ 8192 byte payload packets (4096 channels, 64 channels per destination)

8 bit mode:
128 destinations w/ 8192 byte payload packets (4096 channels, 32 channels per destination)

Stretch goal 2:
Pol, Channel, Time -> Pol, Time, Channel ordering option

realtimeradio, 2020-10-12:
Packet format we have now is what is described here (fastest to slowest: pol, chan, time).
Dave requested fastest to slowest pol, time, chan. You definitely want the thing written here? :)

siemion, 2020-10-12:
If it fits in the BRAM envelope, pol:time:chan (a big chunk of chan 0, a big chunk of chan 1, etc).

siemion, 2020-10-12:
Also per https://github.com/realtimeradio/ata_snap/issues/9 - up to 4096 sample delay would be useful if it fits, but how to prioritize this vs. packet reordering isn't clear (as both will put pressure on BRAM usage)

realtimeradio, 2020-10-16:
@david-macmahon @wfarah 

The multiple packet formats, and dave's swizzle formatting make me want to add fields for "is in swizzled mode" and "n_chans, n_times, n_pols". As a result, I'm going to boost the header to 128 bits, and create these. With the extra space, I'll up the timestamp to 64 bits.
I'll update the docs as soon as I'm convinced the new packetizer works.