realtimeradio/ata_snap #15: Use SNAP board's 1pps rather than ADC's

wfarah, 2020-11-06:
As already noticed before, the ADC 1pps is dropping pulses. We changed the wiring such that we use the 1pps directly to the SNAP board, and it seems that it is working as expected with no pulses seen to be dropping.

A change of value to the `sync_sel` register must be performed somewhere to reflect this change and make it a default. We wired all the 1pps here accordingly

realtimeradio, 2020-11-10:
Do you want this to be --
- A config file parameter?
- An initialization script command line parameter?
- fixed so you never have to think about it?

wfarah, 2020-11-10:
Fixed. Until we find a critical flaw in using the connection

realtimeradio, 2020-11-10:
Should now be fixed in 
fee15a2

closed 2020-11-15 by realtimeradio

wfarah, 2020-11-15:
A tiny bug in:

https://github.com/realtimeradio/ata_snap/blob/11c34b25ceb1ea36d18c681b723524102bdb86fe/sw/ata_snap/src/ata_snap_fengine.py#L141

`_` in `_pps_source` needs to be removed