realtimeradio/ata_snap #24: AtaSnapFengine._read_headers bug

RocketRoss, 2021-03-01:
[ata_snap_feng:L1518](https://github.com/realtimeradio/ata_snap/blob/main/sw/ata_snap/src/ata_snap_fengine.py#L1518) `ips = struct.unpack('>%dI' % n_words, ips_raw)` complains that `ips_raw` isn't correctly sized, which is because it is of length 8*n_words instead of 4*n_words.

Fix is on [L1516](https://github.com/realtimeradio/ata_snap/blob/main/sw/ata_snap/src/ata_snap_fengine.py#L1516).
Will make a pull request towards this end.

RocketRoss, 2021-03-01:
...and 
[L1516](https://github.com/realtimeradio/ata_snap/blob/main/sw/ata_snap/src/ata_snap_fengine.py#L1516) should rather read `'packetizer%d_ips'`

wfarah, 2021-04-21:
@realtimeradio

PR #25 (2021-04-26): ^~ ips_raw fpga.read
Fixes #24
closed 2021-04-26 by realtimeradio